ccrause/fpc-esp-freertos #1: Use by fpcupdeluxe

LongDirtyAnimAlf, 2022-01-07:
A request.
As asked by fpcupdeluxe users, I would like to use these tools to setup cross-compiling to esp8266.
However.
Up needs a fixed scheme to be able to find and install these tools. At the moment, using a snapshot system is not ideal.
So, question.
Are you planning to use a release system for these tools ? That would be very helpful.
Thanks.

ccrause, 2022-01-07:
My snapshots are basically what I generate while testing and debugging the build process of the esp SDKs so should perhaps be seen as test versions. I don't mind changing to a release scheme, but perhaps we should rather update https://github.com/michael-ring/espsdk4fpc with the latest options as required.  

If you are satisfied with the presentation of the builds in espsdk4fpc I can tweak the script to generate the latest stable versions and ask Michael to load the new release versions (it needs updating anyway to include some linker scripts to help with the user compilation step.).  Note that espsdk4fpc bundles both the esp32 and esp8266 tools and libraries already.

Anyway, I'm willing to support whatever option is the best fit with existing fpcupdeluxe functionality.

LongDirtyAnimAlf, 2022-01-07:
Thank you. And thanks for explaining. As far as fpcupdeluxe is concerned:
* it needs a reproduceable way to get the (latest) tools.
* it needs a reproduceable way to unpack, install and setup FPC to use these tools.
All other things after that can be handled by a custom tools installer, included into fpcupdeluxe.

michael-ring, 2022-01-07:
Christo, ping me when you have something, it makes sense to put this into espsdk4fpc.

ccrause, 2022-01-08:
There is a merge request with changes to espsdk4fpc to work with esp-idf v4.3.2 and esp8266-rtos-sdk v3.4.

Alfred, not sure if the current combined esp32 and esp8266 archives in espsdk4fpc is best suited to fpcupdeluxe.  Perhaps these should be split into separate esp32 and esp8266 archives?  If so, it should probably be requested on Michael's espsdk4fpc repository.

michael-ring, 2022-01-18:
I guess we can close this issue, all is prepared now and seems to work, only some twaks needed by Alfred

LongDirtyAnimAlf, 2022-01-19:
Thanks for everything ! But please do not close yet:  at the moment, I do not know how to proceed.
Fpcupdeluxe uses the tools : https://github.com/michael-ring/espsdk4fpc/releases
Are these to be abandoned ?
And which tools does fpcupdeluxe need to use ? From which repo ?
Please point me precisely towards the tools and the implementation.
As any software, fpcupdeluxe needs a precise flowchart of commands to be able to install and use the new tools.
And I am not yet able to extract these commands out of our communications.
Sorry about this.

michael-ring, 2022-01-19:
https://github.com/michael-ring/espsdk4fpc/releases was updated to latest code from Christo.
There is no need to change anything on your side except 

https://github.com/LongDirtyAnimAlf/fpcupdeluxe/issues/487

Alfred, did you receive mails from me over the last days?

LongDirtyAnimAlf, 2022-01-19:
> Alfred, did you receive mails from me over the last days?

Yes ! But I am very busy, also with programming, so my concentration-span for fpupdeluxe is limited.
That is why I am in need of more (perhaps trivial) guidance than normally.

ccrause, 2022-01-19:
Hi Alfred.  No worries, I have been wondering how you managed to invest so much time and effort into figuring out all the install details of the many different cross combinations.  I am relieved that you are showing signs of being human with real life time constraints!

I can try to patch fpcupdeluxe for esp8266, but it will take some time (real life is also distracting me a bit).

LongDirtyAnimAlf, 2022-01-19:
Please hold your effort for now.
I am implementing the changes right now. Should be ready in a couple of hours for you to test.
Thanks anyway !

LongDirtyAnimAlf, 2022-01-19:
Please test:
https://github.com/LongDirtyAnimAlf/fpcupdeluxe/releases/tag/v2.2.0i

ccrause, 2024-08-29:
Closing, issue is related to fpcupdeluxe and espsdk4fpc projects.

closed 2024-08-29 by ccrause