ccrause/fpc-esp-freertos #2: Hardcoded defaults in lpi files cause build fail

michael-ring, 2022-01-10:
I am trying to build the blink example 

~/fpcupdeluxe/lazarus/lazbuild blink.lpi

but the build fails for two reasons:
fpc compiler is hardcoded to /home/christo/fpc/3.3.1/compiler/ppcrossxtensa

it should be either ppcrossxtensa, or even better, not be defined at all and default fpc should choose crosscompiler by flags provided.

Then you refer to a private fpc.cfg file that is not shared in this repo

Fatal: (11009) Unable to open file ~/fpc/3.3.1/fpc.cfg

is it necessary to use this fpc.cfg? For what reasons? How much of this can end up in the lpi file?

ccrause, 2022-01-10:
Yes, not ideal.  However I do not want to hide all the different folders that needs to be searched by the compiler and linker.  This is often where the difficulty lies in getting a new setup to compile and link.  The SDK snapshots in this repository has a different layout compared to espsdk4fpc, and following the steps in https://wiki.freepascal.org/Xtensa will result in yet another folder structure.

Perhaps I can give a description of what some typical configurations may look like, but this will require some work by a user to setup his/her system.

Anyway, I'm quite disorganized.  Do you have a suggestion to make the .lpi files more universal (i.e. less prone to lead to failures on first compile) and work with different configurations of bintools and compiler folders?

michael-ring, 2022-01-10:
I had pretty good working versions for the lpi’s in the past, once compilation works again for me on the new sdk I can try to come up with something hopefully consistent.

I have not done a lot of freepascal in the last months so i need to re-activate some braincells again to remember how things work.

michael-ring, 2022-01-18:
Created pull request.

closed 2022-01-20 by michael-ring