ccrause/fpc-esp-freertos #3: Linking fails for blink example

michael-ring, 2022-01-10:
I am compiling default target (esp32) of blink example but get a linker error that refers esp8266rtos. Any ideas?
I have also manuall fixed issues in fpc.cfg (see https://github.com/LongDirtyAnimAlf/fpcupdeluxe/issues/487 )

```
FPC=fpc.sh lazbuild blink.lpi 
using config file /home/ring/fpcupdeluxe/lazarus/lazarus.cfg
SetPrimaryConfigPath NewValue="/home/ring/fpcupdeluxe/config_lazarus" -> "/home/ring/fpcupdeluxe/config_lazarus"
Hint: (lazarus) [RunTool] "/home/ring/fpcupdeluxe/fpc/bin/x86_64-linux/fpc.sh" "-iWTOTP"
Hint: (lazarus) [RunTool] "/home/ring/fpcupdeluxe/fpc/bin/x86_64-linux/fpc.sh" "-va" "compilertest.pas"
Hint: (lazarus) [RunTool] "/home/ring/fpcupdeluxe/fpc/bin/x86_64-linux/fpc.sh" "-iWTOTP" "-Pxtensa" "-Tfreertos"
Hint: (lazarus) [RunTool] "/home/ring/fpcupdeluxe/fpc/bin/x86_64-linux/fpc.sh" "-va" "compilertest.pas" "-Pxtensa" "-Tfreertos"
Hint: (11030) Start of reading config file /home/ring/fpcupdeluxe/fpc/bin/x86_64-linux/fpc.cfg
Hint: (11031) End of reading config file /home/ring/fpcupdeluxe/fpc/bin/x86_64-linux/fpc.cfg
Free Pascal Compiler version 3.3.1-9888-g9d95e6159e [2022/01/09] for xtensa
Copyright (c) 1993-2021 by Florian Klaempfl and others
(1002) Target OS: FreeRTOS
(3104) Compiling blink.pp
Fatal: (10022) Can't find unit esp8266rtos_30300 used by blink
Fatal: (1018) Compilation aborted
Error: /home/ring/fpcupdeluxe/fpc/bin/x86_64-linux/ppcrossxtensa returned an error exitcode
Error: (lazarus) Compile Project, Mode: ESP32, OS: freertos, CPU: xtensa, Target: /home/ring/devel/fpc-esp-freertos/examples/blink/bin/blink: stopped with exit code 1
Error: (lazbuild) failed compiling of project /home/ring/devel/fpc-esp-freertos/examples/blink/blink.lpi

```

Content of blink.compiled does not look too bad:

<?xml version="1.0" encoding="UTF-8"?>
<CONFIG>
  <Compiler Value="/home/ring/fpcupdeluxe/fpc/bin/x86_64-linux/fpc.sh" Date="1641817452"/>
  <Params Value=" -Tfreertos -Pxtensa -CpLX6 -MObjFPC -Scghi -O1 -gw3 -l -vewnhibq -Fi/home/ring/devel/fpc-esp-freertos/examples/blink/lib/xtensa-freertos -Fi/home/ring/devel/fpc-esp-freertos/freertos-fpc -Fi/home/ring/devel/fpc-esp-freertos/freertos-fpc/esp-idf -Fu/home/ring/devel/fpc-esp-freertos/freertos-fpc -Fu/home/ring/devel/fpc-esp-freertos/freertos-fpc/esp-idf -Fu/home/ring/devel/fpc-esp-freertos/examples/blink/ -FU/home/ring/devel/fpc-esp-freertos/examples/blink/lib/xtensa-freertos/ -FE/home/ring/devel/fpc-esp-freertos/examples/blink/bin/ -o/home/ring/devel/fpc-esp-freertos/examples/blink/bin/blink -Wpesp32 blink.pp"/>
  <Complete Value="False"/>
</CONFIG>

Also, here's my blink.lpi:

<?xml version="1.0" encoding="UTF-8"?>
<CONFIG>
  <ProjectOptions>
    <Version Value="12"/>
    <General>
      <SessionStorage Value="InProjectDir"/>
      <Title Value="blink"/>
      <UseAppBundle Value="False"/>
      <ResourceType Value="res"/>
    </General>
    <BuildModes Count="2">
      <Item1 Name="ESP32" Default="True"/>
      <Item2 Name="ESP8266">
        <CompilerOptions>
          <Version Value="11"/>
          <Target>
            <Filename Value="bin/blink"/>
          </Target>
          <SearchPaths>
            <IncludeFiles Value="$(ProjOutDir);../../freertos-fpc;../../freertos-fpc/esp8266-idf"/>
            <OtherUnitFiles Value="../../freertos-fpc;../../freertos-fpc/esp8266-idf"/>
            <UnitOutputDirectory Value="lib/$(TargetCPU)-$(TargetOS)"/>
          </SearchPaths>
          <CodeGeneration>
            <TargetProcessor Value="lx106"/>
            <TargetCPU Value="xtensa"/>
            <TargetOS Value="freertos"/>
          </CodeGeneration>
          <Linking>
            <Debugging>
              <DebugInfoType Value="dsDwarf3"/>
              <UseLineInfoUnit Value="False"/>
            </Debugging>
          </Linking>
          <Other>
            <CustomOptions Value="-Wpesp8266
-Cacall0"/>
          </Other>
        </CompilerOptions>
      </Item2>
    </BuildModes>
    <Units Count="1">
      <Unit0>
        <Filename Value="blink.pp"/>
        <IsPartOfProject Value="True"/>
      </Unit0>
    </Units>
  </ProjectOptions>
  <CompilerOptions>
    <Version Value="11"/>
    <Target>
      <Filename Value="bin/blink"/>
    </Target>
    <SearchPaths>
      <IncludeFiles Value="$(ProjOutDir);../../freertos-fpc;../../freertos-fpc/esp-idf"/>
      <OtherUnitFiles Value="../../freertos-fpc;../../freertos-fpc/esp-idf"/>
      <UnitOutputDirectory Value="lib/$(TargetCPU)-$(TargetOS)"/>
    </SearchPaths>
    <CodeGeneration>
      <TargetProcessor Value="lx6"/>
      <TargetCPU Value="xtensa"/>
      <TargetOS Value="freertos"/>
    </CodeGeneration>
    <Linking>
      <Debugging>
        <DebugInfoType Value="dsDwarf3"/>
        <UseLineInfoUnit Value="False"/>
      </Debugging>
    </Linking>
    <Other>
      <ConfigFile>
      </ConfigFile>
      <CustomOptions Value="-Wpesp32"/>
    </Other>
  </CompilerOptions>
</CONFIG>

ccrause, 2022-01-10:
Can you give the compiler output in verbose mode (-va).  It seems as if either the subarch or the controller gets changed, perhaps in fpc.cfg?  I cannot see any other reason why an esp8266 related unit would be be searched for.

ccrause, 2022-01-10:
No wait, I can reproduce it. Must be a problem in the compiler.

ccrause, 2022-01-10:
Bug, see https://gitlab.com/freepascal.org/fpc/source/-/merge_requests/137

michael-ring, 2022-01-12:
I can workarround the issue by providing correct -WP.
Actually those are the settings I need specific to LX6:
#IFDEF CPULX6
-Cfhard
-Wpesp32
-WP4.3
-XPxtensa-esp32-elf-
-Ff/home/ring/fpcupdeluxe-qt/cross/bin/xtensa-freertos/esp-idf-4.3.2
-Fl/home/ring/fpcupdeluxe-qt/cross/lib/xtensa-freertos/lx6/
-Fl/home/ring/fpcupdeluxe-qt/cross/bin/xtensa-freertos/esp-idf-4.3.2/components/esp_rom/esp32/ld/
-Fl/home/ring/fpcupdeluxe-qt/cross/bin/xtensa-freertos/esp-idf-4.3.2/components/esp32/ld/
#ENDIF CPULX6

closed 2022-01-12 by michael-ring