ccrause/fpc-esp-freertos #4: Build errors for new 4.3.1 configuration

michael-ring, 2022-01-11:
Making some progress, currently stuck with a define missing, hacked my way by commenting the offending define:
sdkconfig.inc:{ $define CONFIG_FREERTOS_GENERATE_RUN_TIME_STATS := 1}
but this is of course not the solution.

```
clear ; FPC=/home/ring/fpcupdeluxe-qt/fpc/bin/x86_64-linux/fpc.sh /home/ring/fpcupdeluxe-qt/lazarus/lazbuild blink.lpi
using config file /home/ring/fpcupdeluxe-qt/lazarus/lazarus.cfg
SetPrimaryConfigPath NewValue="/home/ring/fpcupdeluxe-qt/config_lazarus" -> "/home/ring/fpcupdeluxe-qt/config_lazarus"
Hint: (11030) Start of reading config file /home/ring/fpcupdeluxe-qt/fpc/bin/x86_64-linux/fpc.cfg
Hint: (11031) End of reading config file /home/ring/fpcupdeluxe-qt/fpc/bin/x86_64-linux/fpc.cfg
Free Pascal Compiler version 3.3.1-9898-gbeecbf1581 [2022/01/10] for xtensa
Copyright (c) 1993-2021 by Florian Klaempfl and others
(1002) Target OS: FreeRTOS
(3104) Compiling blink.lpr
(3104) Compiling /home/ring/devel/fpc-esp-freertos/freertos-fpc/esp-idf/freertos.pp
/home/ring/devel/fpc-esp-freertos/freertos-fpc/esp-idf/freertos.pp(142,2) Error: (2023) User defined: If configGENERATE_RUN_TIME_STATS is defined then portCONFIGURE_TIMER_FOR_RUN_TIME_STATS must also be defined.  portCONFIGURE_TIMER_FOR_RUN_TIME_STATS should call a port layer function to setup a peripheral timer/counter that can then be used as the run ti
/home/ring/devel/fpc-esp-freertos/freertos-fpc/esp-idf/freertos.pp(147,2) Error: (2023) User defined: If configGENERATE_RUN_TIME_STATS is defined then either portGET_RUN_TIME_COUNTER_VALUE or portALT_GET_RUN_TIME_COUNTER_VALUE must also be defined.  See the examples provided and the FreeRTOS web site for more information.
/home/ring/devel/fpc-esp-freertos/freertos-fpc/esp-idf/freertos.pp(303,1) Fatal: (10026) There were 2 errors compiling module, stopping
Fatal: (1018) Compilation aborted
Error: /home/ring/fpcupdeluxe-qt/fpc/bin/x86_64-linux/ppcrossxtensa returned an error exitcode
Error: (lazarus) Compile Project, OS: freertos, CPU: xtensa, Target: blink: stopped with exit code 1
Error: (lazbuild) failed compiling of project /home/ring/devel/blink/blink.lpi
```

ccrause, 2022-01-11:
Can you attach or email the sdkconfig generated by the esp-idf example used to build the libraries?  The sdkconfic.inc file is a Pascal translation of the sdkconfig file to ensure that the Pascal functionality is in sync with the configuration of the precompiled esp-idf libraries.

The current state with a sdkconfig.inc file that is not guaranteed to be in sync with the precompiled libraries is error prone, so a better alternative may be to generate the sdkconfig.inc when buidling the esp-idf libraries.  A bit more complicated to make this framework interact with externally supplied libraries and configuration, but it will avoid his situation.

michael-ring, 2022-01-11:
I am using the version that you provided in 

freertos-fpc/esp-idf/sdkconfig.inc

I try to survive without using any config (except path's) done by me

ccrause, 2022-01-11:
Apologies, it was all due to my laziness. The missing file with the required definitions have been added in 5f7265af396a.

michael-ring, 2022-01-12:
looking pretty good now, the only thing missing now seems to be that I do not set the esp_idf directory and for that reason calline esptool.py fails. I am checking now if an option exists to override/set the directory from fpc commandline.

ccrause, 2022-01-12:
You can specify idf_path using the -Ff command line option[1].  This option overrides the IDF_PATH environment variable [2].

1. https://gitlab.com/freepascal.org/fpc/source/-/blob/main/compiler/options.pas#L2051
2. https://gitlab.com/freepascal.org/fpc/source/-/blob/main/compiler/systems/t_freertos.pas#L1242

michael-ring, 2022-01-12:
Building now seems to work, thank you for your assistance

closed 2022-01-12 by michael-ring