ccrause/fpc-esp-freertos #7: Fatal: Can't find unit system used by adcesp8266test

limuzi2012, 2023-03-13:
编译工程,模式: ESP8266,操作系统:freertos, CPU: xtensa,目标:D:\git\mystudy\2023-3-13 lazarus ESP32 ESP8266\fpc-esp-freertos-master\examples\adc8266test\bin\adc8266test: 退出代码1,错误: 1,提示:2
Hint: Start of reading config file G:\fpcupdeluxe\fpc\bin\x86_64-win64\fpc.cfg
Hint: End of reading config file G:\fpcupdeluxe\fpc\bin\x86_64-win64\fpc.cfg
Verbose: Free Pascal Compiler version 3.3.1-12543-g77f53ebde3 [2023/03/08] for xtensa
Verbose: Copyright (c) 1993-2023 by Florian Klaempfl and others
Verbose: Target OS: FreeRTOS
Verbose: Compiling adc8266test.pp
Verbose: PPU Loading G:\fpcupdeluxe\fpc\units\xtensa-freertos\system.ppu
Fatal: Can't find unit system used by adcesp8266test
Verbose: Compilation aborted
Verbose: G:\fpcupdeluxe\fpc\bin\x86_64-win64\ppcrossxtensa.exe returned an error exitcode

ccrause, 2023-03-13:
`Can't find unit system` typically means there is either a problem with the fpc.cfg file not pointing to the correct RTL unit folder, or the RTL units for the LX106 subarch is not installed.  Can you check that the following folder exists: G:\fpcupdeluxe\fpc\units\xtensa-freertos\lx106

If this folder does not exist you have to select the lx106 subarch.  In fpcupdeluxe, go to the "cross" tab, select "xtensa" and "freertos, then click on the "Set subarch" button and select "lx106".  Now click the "Install cross-compiler".

The xtensa-freertos target has not been tested much on Windows, so there may be more problems.  Please report them so that the Windows experience vcan be improved.

ccrause, 2024-08-29:
No feedback.

closed 2024-08-29 by ccrause