ccrause/fpc-esp-freertos #9: ESP32    wifi initialization failed

yoozaa80, 2024-02-21:
fpc-esp-freertos\examples\wifiscan
fpc-esp-freertos\examples\simplehttpserver

same error

`esp_wifi_init
E (545) wifi:invalid magic number 4008119b, call WIFI_INIT_CONFIG_DEFAULT to init config
I (545) wifi_init: rx ba win: 6
I (545) wifi_init: tcpip mbox: 32
I (555) wifi_init: udp mbox: 6
I (555) wifi_init: tcp mbox: 6
I (555) wifi_init: tcp tx win: 5744
I (565) wifi_init: tcp rx win: 5744
I (565) wifi_init: tcp mss: 1440
I (575) wifi_init: WiFi IRAM OP enabled
I (575) wifi_init: WiFi RX IRAM OP enabled
Error: ESP_ERR_INVALID_ARG
esp_wifi_set_mode
Error: ESP_ERR_WIFI_NOT_INIT
esp_wifi_start
Error: ESP_ERR_WIFI_NOT_INIT
Error: ESP_ERR_WIFI_NOT_INIT
Error: ESP_ERR_WIFI_NOT_INIT
Error: ESP_ERR_WIFI_NOT_INIT
AP #0: .@
  bssid: 3C 13 08 40 9B 11 
  primary: 0
  secondary: WIFI_SECOND_CHAN_NONE
  rssi: 0
  AuthMode: Runtime error 107 at $400D0FB2
  $400D0FB2
  $400D1011
  $400D26BA
  $40140039
  $4008BAD5
`
Whether to provide a full compilation environment?

ccrause, 2024-02-21:
Confirmed. Will investigate further.

ccrause, 2024-02-22:
Please check if the latest update fixes your problem.  The layout of some records changes between esp-idf versions and the latest update adjusts the layout based on the version specified with the -WP compiler option.  To ensure the correct record layout is selected, please pass -WP4.3.1 to the compiler together with using the latest snapshot fpc-esp-idf-4.3.1-20240222.

yoozaa80, 2024-02-22:
Yes!!   wifi init fine！

Enter scan
I (580) wifi:wifi driver task: 3ffc28f4, prio:23, stack:6656, core=0
I (580) system_api: Base MAC address is not set
I (580) system_api: read default base MAC address from EFUSE
I (590) wifi:wifi firmware version: 88c8747
I (600) wifi:wifi certification version: v7.0
I (600) wifi:config NVS flash: enabled
I (600) wifi:config nano formating: disabled
I (600) wifi:Init data frame dynamic rx buffer num: 32
I (600) wifi:Init management frame dynamic rx buffer num: 32
I (610) wifi:Init management short buffer num: 32
I (610) wifi:Init dynamic tx buffer num: 32
I (620) wifi:Init static rx buffer size: 1600
I (620) wifi:Init static rx buffer num: 10
I (630) wifi:Init dynamic rx buffer num: 32
I (630) wifi_init: rx ba win: 6
I (630) wifi_init: tcpip mbox: 32
I (640) wifi_init: udp mbox: 6
I (640) wifi_init: tcp mbox: 6
I (650) wifi_init: tcp tx win: 5744
I (650) wifi_init: tcp rx win: 5744
I (650) wifi_init: tcp mss: 1440
I (660) wifi_init: WiFi IRAM OP enabled
I (660) wifi_init: WiFi RX IRAM OP enabled
I (670) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
W (830) phy_init: saving new calibration data because of checksum failure, mode(0)
I (850) wifi:mode : sta (f0:08:d1:f2:7f:04)
I (850) wifi:enable tsf
AP #0: YOOZAA2G
  bssid: 40 A5 EF 68 B6 2A 
  primary: 1
  secondary: WIFI_SECOND_CHAN_NONE

------------------------------------------------------------
By the way, how do I compile libesp32.a, I want to try the new version later！  Thanks！

ccrause, 2024-02-22:
> By the way, how do I compile libesp32.a, I want to try the new version later！

The process is briefly described [on the FPC wiki](https://wiki.freepascal.org/Xtensa#Preparations).
Another option is to look at [the esptool.py script](https://github.com/michael-ring/espsdk4fpc/blob/main/esptool.py) used to build the libraries for fpcupdeluxe.

Please let me know if you managed to build and link a FPC project using esp-idf v5. My tests so far failed with the esp32 linker not being able to locate libraries, despite specifying the correct locations.