ccrause/fpc-esp-freertos #10: undefined reference to `uxTaskGetSystemState'

knakkie, 2024-08-13:
I am very happy with the possibility of programming ESP32 with Pascal!
While compiling the example: “realtimestats´ I get the error message:
(45,0) Error: linker: undefined reference to `uxTaskGetSystemState'
realtimestats.pp(63,0) Error: linker: undefined reference to `uxTaskGetSystemState'
How can I solve this error?

ccrause, 2024-08-14:
I do most of my development and testing on Linux, so it is possible that the Windows snapshot was built with the wrong config.  To use the `uxTaskGetSystemState` function requires enabling the `CONFIG_FREERTOS_GENERATE_RUN_TIME_STATS` option in the esp-idf config.

This option is not enabled in the Windows build, which is unfortunate!  As a work-around, you can replace the files in **lib/xtensa-freertos/lx6/** with the contents of the **libs/** folder in the [Linux snapshot](https://github.com/ccrause/fpc-esp-freertos/blob/master/snapshot/fpc-esp-idf-4.3.1-20240222.zip).

If this works, I can update the Windows snapshot with the Linux libs, else I have to rebuild the Windows snapshot which will take some time.

knakkie, 2024-08-14:
Thank you, that was the solution!
I have replaced the files in the lib/xtensa-freertos/lx6/ with the files from the Linux folder.
There are more files in the lib/xtensa-freertos/lx6/ then in the Linux folder so I keep the files that are not in the Linux folder.

ccrause, 2024-08-24:
Please test the new Windows [snapshot](https://github.com/ccrause/fpc-esp-freertos/blob/master/snapshot/esp-idf-4.3.7-win64-20240822.zip).  If it works this issue can be closed.

knakkie, 2024-09-01:
Sorry for the late reply.
I have installed the esp-idf-4.3. 7-win64 Snapshot.
When I compile/build the Blink example I get the following error messages:
Debug: C:\fpcupdeluxe_Esp32_522\cross\bin\xtensa-freertos\bin\xtensa-esp32-elf-ld.exe: C:\fpcupdeluxe_Esp32\fpc-esp-freertos-master\examples\blink\bin\blink.elf section `.data' will not fit in region `dummy'
Debug: C:\fpcupdeluxe_Esp32_522\cross\bin\xtensa-freertos\bin\xtensa-esp32-elf-ld.exe: region `dummy' overflowed by 20 bytes

ccrause, 2024-09-02:
> Debug: C:\fpcupdeluxe_Esp32_522\cross\bin\xtensa-freertos\bin\xtensa-esp32-elf-ld.exe: C:\fpcupdeluxe_Esp32\fpc-esp-freertos-master\examples\blink\bin\blink.elf section `.data' will not fit in [...]

This should be fixed by commit [d53a81e2](https://gitlab.com/freepascal.org/fpc/source/-/commit/d53a81e2d96bd580dcc90fe32fdf3d71cb412b9c) in FPC, but I haven't yet tested this on Windows.

knakkie, 2024-09-02:
In the file "readme.txt" of the snapshot 4.3.7 you have to use "-WP4.3.3" in the "fpc.cfg" file, I changed this to "-WP4.3.7" and now the error message is gone.
The "realtimestats" program I can now compile/build without problems

ccrause, 2024-09-03:
Thank you for reporting the mistake in the readme, fixed in commit fb5ddca.

You are welcome to discuss your experience using FPC and ESP32 (or ask more questions) in the [embedded](https://forum.lazarus.freepascal.org/index.php?board=77.0) board on the Lazarus forum.

closed 2024-09-03 by ccrause